isoverse/isoreader #3: support remote file system

sebkopf, 2017-10-23:
potentially with google drive and/or dropbox
for google drive use : http://googledrive.tidyverse.org/

japhir, 2018-07-05:
It would also be nice if it would just work with samba-mounted drives, i.e. allow "smb://path/to/samba" file locations.

sebkopf, 2018-11-13:
agreed, unfortunately internal file reading functions in R only support local files, ftp and http and not the samba protocol. That said, it might work with curl along these lines although not quite clear how to best handle required passwords and usernames. This will be easier to implement once #10 is resolved since the root path parameter could allow for remote file objects (`RemoteSMB`, `RemoteGDrive`, `RemoteDropbox`, etc.) that store login and other credentials in addition to the location.

```
handle <- new_handle()
handle_setopt(handle, username = "domain\\username")
handle_setopt(handle, password = "secret") # If needed
request <- curl_fetch_memory("smb://host.example.com/share/file.txt", handle = handle)
content <- rawToChar(request$content)
```