isoverse/isoreader #5: include file_info in the data frame used for plotting

sebkopf, 2017-10-25:
this will make it possible to apply facets and other parameters on top of the base chromatogram

japhir, 2018-11-09:
For my analysis I simply use:

```{r}
  dids <- iso_read_dual_inlet("/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData/253pluskiel/Raw Data/Kiel Raw Data",
    discard_duplicates = FALSE)

  info  <- dids %>%
    iso_get_file_info()

  rawdata  <- dids %>%
    iso_get_raw_data()

  flatdata  <- full_join(info, rawdata, by = "file_id") %>%
    group_by(file_id) %>%
    arrange(file_id, type, cycle) %>%
    mutate(broadid = ifelse(`Identifier 1` %in% paste0("ETH-", 1:4),
      `Identifier 1`, "other"))

  rm(info, rawdata)
```
and work from there, so that every line always has the fileinfo attached. Is there a better way?

sebkopf, 2018-11-13:
yes, thanks! I forgot to close it. `include_file_info` is indeed the best way to include the info.

closed 2018-11-13 by sebkopf