isoverse/isoreader #7: improve read speed

sebkopf, 2017-11-21:
binary parsing takes a long time but should be possible to improve in speed by the following means:
 - consider moving key processing steps into C++
 - consider parallelizing processing tasks that can run in parallel

sebkopf, 2018-11-13:
unfortunately the problem is the slow processing of the binary files with raw regular expressions rather than the file loading itself (the isodat files especially are actually pretty small overall, just a few 100 kb). 

would be worthwhile revisiting a move to C++ at some point also because it would provide an easy way to implement a direct python API but for now, parallelization will have to suffice

the new parameter is called `parallel = TRUE` and is now part of the dev branch (will be in v 1.0)

japhir, 2018-12-05:
Currently, on the dev branch I get this error when reading files with the `parallel` option:
```{r}
Progress: [------------------------------------------------] 0/5090 (  0%)  0sError in stri_replace_all_fixed(string, pattern, replacement, vectorize_all = vec,  : 
  object 'msg' not found
```
So a few things:

- looks like the progress bar doesn't play nice with errors
- the msg that you tried to fix in the latest commit doesn't seem to have worked

Cheers! :)

sebkopf, 2018-12-07:
hi @japhir - this is very strange, I thought I had fixed this bug. Do you have the package loaded in quick memory via `devtools::load_all()` or actual reinstllation? The only thing I can think of is that somehow an older version of isoreader is still somewhere in memory and when the parallel processes dispatch the call to a new session it loads the old version. Could you do me a favor and check what happens if you uninstall isoreader and then reinstall/build from `dev` branch?

japhir, 2018-12-07:
Oh shit, it stopped working again:
```
Progress: [============================================>] 5070/5118 ( 99%)  1mError in base::nchar(strip_style(x), ...) : 
  invalid multibyte string, element 1
```
But when I just ran it again, it did work without error… Peculiar.

Unfortunately, running the code takes quite long, I think mostly because recursively finding the did/caf files in the remote samba drive takes ages.

sebkopf, 2018-12-07:
mh, those look like errors that happen during the reading of the log file (which is outside the error catching of isoread because it's in the parent process). I suspect the log file can get corrupt every once in a while with all the different processes writing to the same file. I added a safety check in the last commit just now that catches any errors that come from reading the log file and resets it (with an info message) in case there is trouble so things can progress without uncaught trouble. These are tricky to reproduce but if you see them again without it being caught nicely, please let me know.

regarding the `future` package,  it's in the dependency list so I am surprised `isntall_github()` didn't install it automatically, very strange (this problem existed on windows before but I thought it had been fixed in devtools)

I think you are right that you're not supposed to load source packages that way.

japhir, 2020-02-17:
Today it threw this at me again:
```
Progress: [============================>---] 12847/14112 ( 91%)  2mError in base::nchar(strip_style(x), ...) : 
  invalid multibyte string, element 1
```

sebkopf, 2020-02-17:
the problem with these multibyte errors is that they have nothing to do with isoreader but with how some output function (here probably the progress bar calling base::nchar) handles unicode characters (not well apparently in this case). could you send me the example file in question (#12847) and let me know your current isoreader version? (1.1.0 or older?)

sebkopf, 2020-03-01:
hi @japhir : I implemented a few more safety checks and hopefully more useful error messages in the text processing base functions - can you test and see if you get a more useful error?

japhir, 2020-04-01:
Hey! I just tested this because I had to re-read my entire database (see other issue) and this error seems resolved now!

sebkopf, 2020-04-01:
great, thanks!

closed 2020-04-01 by sebkopf