isoverse/isoreader #10: saving file paths in isoread

sebkopf, 2017-12-02:
a few improvements to the way iso_files store and use file paths to make `reread` more portable between server and local file systems (and of course between different operating systems)

- include option to save only the relative file paths (relative to working directory) if users pass in file paths that include the working directory - make this the default behavior so file system information is not stored in the $file_info$file_path and it is easier to re-read files even after moving around directories

- perhaps also store information about whether the file path is absolute or relative

- note: this should probably go hand in hand with functions to replace parts of the file path for entire iso_file collections

japhir, 2018-11-09:
This would make reading from a (terribly slow) remote much easier. I think reading the actual files from the local cache takes very little time, but searching for all the .did files in the remote, and then looking them up takes ages.
So ideally it would store a list of files, then check if there are any new ones (filter by date previously checked for increased speed? Not sure if possible).

sebkopf, 2019-03-04:
file path storage implemented, remote file system is separate issue #3

closed 2019-03-04 by sebkopf