isoverse/isoreader #11: allow easy recoding of data columns

sebkopf, 2017-12-02:
in order to co-process data from different file formats, it would be very useful to have a "rename" function that recodes multiple columns into one (making sure that they are all NA in only one set of files).

Pseudo-syntax suggestion:
- `iso_recode(name = c(Id1, info), preparation = c(Comment, Preparation))`

japhir, 2018-11-07:
You could use [dplyr's unite](https://tidyr.tidyverse.org/reference/unite.html) for this.

sebkopf, 2018-11-08:
good idea, I think what unite doesn't do well though is deal with `NA` - I imagine a scenario where you have e.g. info `x` from one type of IRMS format and info `y` from another but you want to work with them as a single `id` column: `data_frame(x=c(NA, "a"), y = c("b", NA)) %>% unite(id, c(x, y))` creates `NA_b` and `a_NA`. 

Maybe just a `case_when` would be easiest, that way the order the columns are provided also determines order of precedence.  Maybe something like this with x, y, z dynamically implemented using the `unite` syntax?

```r
data_frame(
 x=c(NA, "xb", "xc"), 
 y = c("ya", NA, NA), 
 z = c("za", "zb", "zc")) %>% 
mutate(
 id = case_when(
  !is.na(x) ~ x, 
  !is.na(y) ~ y, 
  !is.na(z) ~ z, 
  TRUE ~ NA_character_))
```

japhir, 2018-11-08:
Hmm in that case I don't understand what you want to achieve. This would always pick the first that has a non-na value and ignore further columns, right?

```{r}
# A tibble: 3 x 4
  x     y     z     id   
  <chr> <chr> <chr> <chr>
1 NA    ya    za    ya   
2 xb    NA    zb    xb   
3 xc    NA    zc    xc   
```

sebkopf, 2019-03-23:
actually implemented this with regular dplyr rename but executed for each isofile separately so recoding to the same column name is easy

sebkopf, 2019-03-26:
fully implemented in dev version with `iso_rename_file_info()` and `iso_select_file_info()` and supported in all `iso_get...` functions.

commit b53aad75 (2019-03-24): implement rename functionality

closes #11
closed 2019-03-26 by sebkopf