isoverse/isoreader #23: updates on .caf reader

japhir, 2018-07-05:
Hi Sebastian,

Finally coming back to taking a look at this (see https://github.com/sebkopf/isoread/issues/21).
Thanks for your elaborate reply in the other thread. I'm sorry I haven't taken the time to look at it before.

The caf reader works really well! It's slow, but I've just read in 564 data files from some MSc-student projects as a test, with only 80 problems in total (mostly relating to failed measurements due to drop tests or similar). I don't know how to throw the errors again, and the printed tibble isn't big enough to show the full extent. Here's a snippet: (... indicates I omitted some successful imports).

```{r}
Warning: caught error - cannot identify measured masses - block 'CResultData' not found after position 1 (pos 109531)
Warning: caught error - cannot process vendor data table - block 'CResultData' not found after position 1 (pos 1)
...
Warning: caught error - cannot identify measured masses - block 'CResultData' not found after position 1 (pos 93743)
Warning: caught error - cannot process vendor data table - block 'CResultData' not found after position 1 (pos 1)
...
Warning: caught error - cannot identify measured masses - block 'CResultData' not found after position 1 (pos 93989)
Warning: caught error - cannot process vendor data table - block 'CResultData' not found after position 1 (pos 1)
...
Warning: caught error - cannot identify measured masses - block 'CResultData' not found after position 1 (pos 109535)
Warning: caught error - cannot process vendor data table - block 'CResultData' not found after position 1 (pos 1)
Info: reading and caching file 563/564 '~/Downloads/megaraw/180522_Stds/180523_Std_ETH-3_17.caf' with '.caf' reader...
Info: reading and caching file 564/564 '~/Downloads/megaraw/180522_Stds/180523_Std_ETH-3_18.caf' with '.caf' reader...
Info: encountered 80 problems in total.
# A tibble: 80 x 4
   file_id                             type  func       details                
   <chr>                               <chr> <chr>      <chr>                  
 1 180306_RobinvdP_MECO_1_1_ETH-1.caf  error extract_c… cannot identify measur…
 2 180306_RobinvdP_MECO_1_1_ETH-1.caf  error extract_c… cannot process vendor …
 3 180306_RobinvdP_MECO_20_2_ETH-3.caf error extract_c… cannot identify measur…
 4 180306_RobinvdP_MECO_20_2_ETH-3.caf error extract_c… cannot process vendor …
...
```

Regarding [your question 2](https://github.com/sebkopf/isoread/issues/21), I think the caf file that I uploaded earlier should be fine!

sebkopf, 2018-11-07:
@japhir could you send me one of those problematic files, maybe `180306_studname_1_1_ETH-1.caf1`? for a full list of errors you can use the `isoreader::get_problems()` function

japhir, 2018-11-08:
Ah, I see now that they are actually failed runs, with a drop test or something similar! The filesizes for both these files are only 335 kb, whereas the other files are all 652.1 kb.

Here are the files nonetheless, in case you'd like to tweak the error messages to be even more informative. (I think "cannot identify measured masses" and "cannot process vendor data table" are fine though).

[empty_caffiles_UU.zip](https://github.com/KopfLab/isoreader/files/2561247/empty_caffiles_UU.zip)

japhir, 2019-12-13:
Hey @sebkopf, I unfortunately have to process some really old caf files. Our old machine has a new Kiel IV now so in the future this won't be a problem again, but for now I think I may have found some parsing bugs for the file_info. I totally forgot about https://github.com/sebkopf/isoread/issues/21, and now see that I was lax in my response to your question about doing more use-case tests. Well, I guess here they are, 745 days later  :blush:.

Here's a list of some problems:
- `file_id`, `file_root`, `file_path` all work as expected. `file_subpath` is always `NA`, but I'm not sure what it should be.
- `Line`: should be either 1 or 2 in my files, but currently has the info that should be in `Row` (a number between 1 and 46).
- `Sample` seems ok (a number between 1 and 24), but is never 1, 2, or 3 and has some `NA`s.
- `Identifier 1` is ok!
- `Identifier 2` sometimes has the `Method` info, other times the `Row` number, other times is `NA` (as it probably should be based on the sequence files for these samples).
- `Analysis` is ok!
- `Comment` seems OK, looks like somebody entered a modified version of `Identifier 1` here somehow, but I guess that's what's in input column of our sequence, so parsing works.
- `Preparation`: only NA's -> this one's the worst, since my script relies on Preparation for the application of the ETF. I can sometimes parse it out of the file_id, but the naming has been very inconsistent in 2017/2018.
- `Pre Script` only NAs (might be the case in the files)
- `Post Script` only NAs (might be the case in the files)
- `Method` is ok
- `Row` should have the info that is presently in `Line` (a number between 1 and 46). At the moment it seems like it's getting `Sample`.
- `Line.1` Duplicate of `Line`, but this one sometimes correctly gets the line number (1 or 2), other times is NA.
- `Peak Center`: only NA's
- No `measurement_info` list column (maybe it's not there? I don't particularly care, since I'm currently not using it).

I don't know how many people will still require a caf reader, but these are some things to fix, probably.

here's how I looked at the columns of interest:

```
for (i in names(caf_file_info)[!names(caf_file_info) %in% c("file_id", "file_datetime", "file_path", "file_subpath", "Analysis", "Identifier 1")]) { message("\n*** NEXT: ", i); caf_file_info %>% pull(i) %>% unique() %>% print()}

*** NEXT: file_root
[1] "/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData/Kiel 253/clumped/Results"

*** NEXT: Line
 [1]  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 20 21 22 24  1  2 19 23 26
[26] 27 25 39 40 41 42 28 29 30 31 32 33 34 35 36 38 43 44 45 46 37

*** NEXT: Peak Center
[1] NA

*** NEXT: Pressadjust
[1] NA

*** NEXT: Background
[1] NA

*** NEXT: Reference Refill
[1] NA

*** NEXT: Weight [mg]
  [1] 163.0 165.0 167.0 154.0 172.0 149.0 120.0 182.0 190.0 164.0 191.0 176.0
 [13] 150.0 168.0 152.0 157.0 171.0 166.0 155.0 159.0 153.0 178.0 156.0 180.0
 [25] 170.0 174.0 179.0 169.0 161.0 160.0 158.0 173.0 162.0 151.0 177.0 175.0
 [37] 139.0 184.0 146.0 138.0 148.0 127.0 143.0    NA 144.0 141.0 136.0 134.0
 [49] 142.0 125.0 147.0 130.0 137.0 129.0 105.0 181.0  87.0 133.0 124.0 121.0
 [61] 189.0 106.0 118.0 140.0  66.0 185.0 131.0 183.0 132.0 128.0  68.0 145.0
 [73] 122.0 123.0 135.0 188.0 110.0  97.0 126.0 115.0 107.0 186.0 198.0  62.0
 [85] 116.0  85.0  69.0 187.0   0.0  88.0 134.7 131.2 119.8 167.3 160.2 182.2
 [97] 179.5 169.9 166.2 123.7 153.5 127.6 138.8 140.1 120.2 164.4 124.9 132.9
[109] 136.8 135.2 162.3 162.6 167.4 161.3 165.9 162.4 174.5 165.7 147.3 194.0
[121] 157.2 162.5 163.8 160.1 119.0 235.0 205.0 211.0 197.0 218.0 196.0  50.0
[133] 114.0 214.0 223.0 199.0 108.0 217.0 206.0 200.0 195.0 210.0 225.0 212.0
[145] 213.0 192.0 233.0 228.0 230.0 227.0 203.0 232.0 250.0 224.0 209.0 113.0
[157] 202.0 109.0 103.0 117.0 101.0 100.0  90.0  98.0 102.0  94.0  84.0 104.0
[169] 112.0 201.0 111.0 243.0  11.0 204.0 193.0  99.0  78.0  93.0  48.0

*** NEXT: Sample
 [1]  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 21 22  2 18 19 20 23 24 NA

*** NEXT: Identifier 2
 [1] NA                                           
 [2] "/"                                          
 [3] "CARBO IIIUvU_Clump_IM44_tt400_refref240.met"
 [4] "CARBO IIIUvU_Clump_IM44_tt360_refref240.met"
[... 59 lines cut ...]
[2] "CARBO IIIUvU_Clump_IM.met"                  
[3] "CARBO IIIUvU_Clump_IM44_tt400_refref240.met"
[4] "CARBO IIIUvU_Clump_IM44_tt360_refref240.met"
[5] "CARBO IIIUvU_Clump_transfert360.met"        
[6] "CARBO IIIUvU_Clump_IM_V25off.met"           
[7] "CARBO IIIUvU_Clump_IM_25offtest.met"        
[8] "Carboclumped_onetab.met"                    
[9] "Getachew4V.met"                             

*** NEXT: MS_integration_time.s
[1] 24 26 18

*** NEXT: Line.1
[1] NA  "1" "2"

*** NEXT: Row
 [1]  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 21 22  2 18 19 20 23 24 NA

*** NEXT: masspec
[1] "Pacman-KielIII"

*** NEXT: broadid
[1] "ETH-2"   "ETH-3"   "ETH-4"   "other"   "ETH-1"   "IAEA-C1" "IAEA-C2"
[8] "Merck"  
```