isoverse/isoreader #24: cannot read non-dual(?) did file

japhir, 2018-07-05:
I guess our new mass spec set up for LIDI-measurements (60 cycles, first the reference gas, then the sample gas) causes the did files to have a different structure from the typical dual-inlet system. Or at least, there is something going on with the "info" capture.

Attached is an example measurement: [link to ETH-1 did file](https://surfdrive.surf.nl/files/index.php/s/Zrss4Tj2zDDuup0)

Reading a few did files (without the cache, so you get the real errors):

```
did  <- iso_read_dual_inlet("raw files/mat253plus/standards/", read_cache = FALSE)
Info: preparing to read 8 data file(s)...
Info: reading and caching file 1/8 'raw files/mat253plus/standards//180515_28_RobinV_1_ETH-1.did' with '.did' reader...
Warning: caught error - cannot process measurement info - 'info' capture failed: raw data does not match requested data pattern (([ -~]\x00){15}): 41 00 63 00 69 00 64 00 3a 00 20 00 37 00 30 00 2e 00 31 00 20 00 5b 00 b0 00 43 00 5d 00
Warning: caught error - cannot process vendor computed data table - unequal number of column headers (10) and data entries (0) found (pos 948896)
Info: reading and caching file 2/8 'raw files/mat253plus/standards//180515_28_RobinV_2_ETH-1.did' with '.did' reader...
Warning: caught error - cannot process measurement info - 'info' capture failed: raw data does not match requested data pattern (([ -~]\x00){16}): 20 00 41 00 63 00 69 00 64 00 3a 00 20 00 36 00 39 00 2e 00 39 00 20 00 5b 00 b0 00 43 00 ...
Warning: caught error - cannot process vendor computed data table - unequal number of column headers (10) and data entries (0) found (pos 949042)
```

Reloading the cache also throws some errors:

```
> did  <- iso_read_dual_inlet("raw files/mat253plus/standards/")
Info: preparing to read 8 data file(s)...
Info: reading file 1/8 'raw files/mat253plus/standards//180515_28_RobinV_1_ETH-1.did' from cache...
Info: reading file 2/8 'raw files/mat253plus/standards//180515_28_RobinV_2_ETH-1.did' from cache...
Info: reading file 3/8 'raw files/mat253plus/standards//180515_28_RobinV_3_ETH-3.did' from cache...
Info: reading file 4/8 'raw files/mat253plus/standards//180515_28_RobinV_4_ETH-3.did' from cache...
Info: reading file 5/8 'raw files/mat253plus/standards//180515_28_RobinV_5_ETH-2.did' from cache...
Info: reading file 6/8 'raw files/mat253plus/standards//180515_28_RobinV_6_ETH-2.did' from cache...
Info: reading file 7/8 'raw files/mat253plus/standards//180515_28_RobinV_7_ETH-3.did' from cache...
Info: reading file 8/8 'raw files/mat253plus/standards//180515_28_RobinV_8_ETH-3.did' from cache...
Info: encountered 17 problems in total.
# A tibble: 17 x 4
   file_id                      type  func                            details  
   <chr>                        <chr> <chr>                           <chr>    
 1 180515_28_RobinV_1_ETH-1.did error extract_isodat_measurement_info "cannot …
 2 180515_28_RobinV_1_ETH-1.did error extract_did_vendor_data_table   cannot p…
 3 180515_28_RobinV_2_ETH-1.did error extract_isodat_measurement_info "cannot …
 4 180515_28_RobinV_2_ETH-1.did error extract_did_vendor_data_table   cannot p…
 5 180515_28_RobinV_3_ETH-3.did error extract_isodat_measurement_info "cannot …
 6 180515_28_RobinV_3_ETH-3.did error extract_did_vendor_data_table   cannot p…
 7 180515_28_RobinV_4_ETH-3.did error extract_isodat_measurement_info "cannot …
 8 180515_28_RobinV_4_ETH-3.did error extract_did_raw_voltage_data    cannot l…
 9 180515_28_RobinV_4_ETH-3.did error extract_did_vendor_data_table   cannot p…
10 180515_28_RobinV_5_ETH-2.did error extract_isodat_measurement_info "cannot …
11 180515_28_RobinV_5_ETH-2.did error extract_did_vendor_data_table   cannot p…
12 180515_28_RobinV_6_ETH-2.did error extract_isodat_measurement_info "cannot …
13 180515_28_RobinV_6_ETH-2.did error extract_did_vendor_data_table   cannot p…
14 180515_28_RobinV_7_ETH-3.did error extract_isodat_measurement_info "cannot …
15 180515_28_RobinV_7_ETH-3.did error extract_did_vendor_data_table   cannot p…
16 180515_28_RobinV_8_ETH-3.did error extract_isodat_measurement_info "cannot …
17 180515_28_RobinV_8_ETH-3.did error extract_did_vendor_data_table   cannot p…
```

sebkopf, 2018-11-07:
@japhir sorry this took a while, there were multiple issues at play in the LIDI files (the high cycle times and several non standard characters in the measurement info), but they should all be solved now

closed 2018-11-07 by sebkopf

japhir, 2018-11-08:
Thanks so much for this! :+1: