isoverse/isoreader #33: Detect duplicate file names with different `file_datetime`s?

japhir, 2018-11-08:
In our database there are several files with the same name (in different subdirectories), that occur because for instance a run has stopped due to some error (e.g. no acid drop recorded). In this case the .did file is still saved, but contains no data except for the diagnostics (line, row, error message, etc.). Because the material is still there, we can restart the run from that particular aliquot, resulting in a new folder with some of the same names.
In our case, over half a year or so this has resulted in 267 files with duplicate names. Would it be possible for isoreader to not stop when importing different files with the same name, if they have different timestamps/file_paths?

japhir, 2018-11-09:
Awesome! Seems to work, but the warnings never mention an added number, i.e.:
```
Warning: duplicate files kept but with recoded file IDs: 181003_96_Ilja_14_IK4-060Sc3.did
```
If I filter the specified file, I do see it.
```{r}
dids <- iso_read_dual_inlet("out.di.rda")
dids %>% iso_filter_files(grepl("^180929_94", file_id))
```
does show the `#1`, `#2`, etc.

japhir, 2020-05-04:
Hey @sebkopf, quick question: does the recoding with new `file_id`'s (with `#1` and `#2` code) check for duplicated file names or for duplicated `Analysis` numbers? Because it looks like in our case we have some fully duplicated files (somebody copied over a folder to some other subfolder for "backup" purposes, very annoying) while there are also duplicated `file_id`'s where a run was stopped, the file was saved with all the errors in it, then the run was restarted, which results in a new file in a different directory with the exact same name (we don't use `Analysis` in the automatic `file_id` creation in isodat).

Right now I'm filtering them out by:

```R
iso_files %>%
     group_by(Analysis) %>%
     add_count() %>%
     filter(n == 1) %>%
     ungroup() %>%
     select(-n)
```

but I'd like to re-recode the `file_id`'s back to without the `#1`.

japhir, 2020-06-02:
Hey @sebkopf, could you answer the above question? I'm still having this issue today, with some files actually duplicated and some files with the same file name but different file contents.

It appears that sometimes even the Analysis number is duplicated in different files in our database :-1:… Could we hash the files without looking at the filename so we can filter out identical files but keep all the others?

japhir, 2020-06-02:
I'm using a workaround that I hope will isolate all the issues like this now:

```{r}
  nodups <- info %>%  # info is the result of `iso_get_file_info()` on my whole record
    # undo unique ID creation
    mutate(file_id = ifelse(str_detect(file_id, "#\\d+$"), str_replace(file_id, "#\\d+$", ""), file_id)) %>%
    # hoping file_datetime and file_size will highlight actual copies in stead of same file_id's
    group_by(Analysis, file_datetime, file_size) %>%
    add_count() %>%  # adds the new n column
    ungroup()

  nodups %>%
    filter(n > 1) %>%
    nrow()
  #> [1] 922

  nodups <- nodups %>%
    tidylog::distinct(Analysis, file_size, file_datetime, .keep_all = TRUE) %>% # I like tidylog to show me how many rows got filtered out. 
    tidylog::select(-n)
  #> distinct: removed 461 rows (3%), 15,078 rows remaining
  #> select: dropped one variable (n)
  # so that's 461 * 2 = 922 duplicated rows, gets rid of all of them!
```

after this, it appears that there are still some files with identical file_id's but different file_size and Analysis numbers, so I did some more tweaking to apply your file_id recoding only to those weird copies.

sebkopf, 2020-06-02:
Hi @japhir, sorry no daycare for kids and a 6 week old at home makes everything slow at the moment.

I think the issue you raise is a very important one and we should find a way to address this directly in the file read commands (although I quite like your work around + tidylog is fantastic, maybe we should use it at some point to replace the isoreader log messages for a more standardized format).

Before I go implementing this I would like to ask for your input if you think this suggestions would work: how about we have two parameters do deal with duplicates instead of the current `discard_duplicates`:
 - `discard_exact_duplicates`: if `TRUE`, anything that has the exact same file hash will only be presented once (first file found). If `FALSE`, uses the existing `#1, #2, #3` nomenclature to deal with the duplicates.  Default `TRUE`?
 - `discard_file_id_duplicates`: if `TRUE`, anything that has the same file id (but not the same hash) will only be presented once (first file found). if `FALSE`, also uses the existing `#1, #2, #3` nomenclature to deal with duplicates. Default `FALSE`?? (I think this is a harder decision). I tend to `FALSE` as the default because it seems these actually represent inherently different data.

Thoughts?

japhir, 2020-06-03:
Hi @sebkopf, no worries!

The proposal sounds great, it would implement what I'm doing with my workaround now. Perhaps the identical duplicated files could still point to both files with file_path/file_id somehow? But that's probably not necessary.

japhir, 2020-10-22:
Hey @sebkopf, did you work on `discard_exact_duplicates` already? This may be a useful feature before uploading to CRAN?