isoverse/isoreader #45: fully implement bgrd_data field

sebkopf, 2019-01-16:
including `iso_convert_bgrds` or similar function to scale backgrounds

sebkopf, 2019-03-27:
@japhir and @brettdavidheiser 
could you guys help me with figuring out what the background can look like in different isodat files? I think it's usually in measurement info but not always the same format. The more examples, the better isoreader can capture the background data right away.

sebkopf, 2019-04-09:
yes, that would be awesome! thanks @japhir

japhir, 2019-04-25:
Waah 16 days ago already! This kept floating around my "Today I'll do this" todo list.

I thought I wouldn't need need to upload the files, if I just:
```{r}
dids %>% 
    iso_get_file_info() %>% 
    pull(measurement_info) %>% 
    map(str_subset, "Background") %>% 
    str_c %>% 
    unique
```
which gives
```
Info: aggregating file info from 7644 data file(s)
[1] "character(0)"                                                               
[2] "Background: BGD 2018/Jan/23 03:15 -  (Administrator)"                         
[3] " Background: BGD 2018/Jan/23 03:15 -  (Administrator)"                        
[4] "Background: BGD 2017/Sep/06 09:00 - ,3.31 mV,3.86 mV,4.54 mV (Administrator)" 
[5] " Background: BGD 2017/Sep/06 09:00 - ,3.31 mV,3.86 mV,4.54 mV (Administrator)"
Warning message:
In stri_c(..., sep = sep, collapse = collapse, ignore_null = TRUE) :
  argument is not an atomic vector; coercing
```
for our 253 plus + Kiel IV setup. (I have to `str_c` in stead of `map_chr`, because some return `character(0)`, and `map` doesn't like that).

for our 253 + Kiel III
```{r}
cafs %>% iso_get_file_info %>% pull(Background) %>% unique
```
```
Info: aggregating file info from 4928 data file(s)
[1] NA
```
And they don't have a `measurement_info` column. Was that the problem you were trying to address?

Attached are some ETH-1's and other random files from a long time ago that should have all of the different combinations of the background value fields. I should still add our 253 + Kiel IV (new setup) did files, but they are currently not accessible remotely.
[files_for_isoreader.zip](https://github.com/isoverse/isoreader/files/3116269/files_for_isoreader.zip)

sebkopf, 2019-04-25:
awesome, thanks for sharing these! yeah, unfortunately I could never find this information in the `caf` files. They don't have `measurement_info` and I did not figure out where else they store this information (if at all). I've mostly given up on the cafs for now, the problem I'm trying to figure out is how variable the background info formatting is between different `.did` files so your collection of different files is super helpful. Is the new setup similar in how it formats background measurement info or about the same?

japhir, 2019-04-25:
Should be the same, we replaced the Kiel III with a Kiel IV, and thus also had to replace the PC and software, so I expect (hope) that the data files are similar, except that maybe we have some different settings in the method.