isoverse/isoreader #64: add a widget function for easy movement around a chromatogram (like in isoviewer)

sebkopf, 2019-03-28:
maybe this should just be functionality available from isoviewer instead? but possible to use inside an rmd file

japhir, 2019-07-10:
I've been creating ggplots and making them interactive using [plotly](https://plotly-r.com/index.html)'s `ggplotly`. This allows me to zoom in on separate axes (draw rectangle, drag borders of specific axis, drag narrow rectangle for one axis only, double click to restore auto-zoom), allows hiding of specific legend entries and is generally great!
One major thing: if you're plotting individual cycles for many many measurements the `geom_point` can get a bit slow (`geom_line` should wourk fine) so you can wrap your can render it using WebGL like so: `toWebGL(ggplotly(dynamicTicks=TRUE))`. The dynamicTicks is so that the tickmarks update whilst zooming in.

sebkopf, 2019-07-10:
I love plotly and I think you're right that isoviewer should provide an easy way to make thinks interactive both with static plots and through ggplotly. The static plot version is nice for zooming in and generating PDFs of subsets of a plot. I think should be easy to just provide both options.