isoverse/isoreader #65: keep file order in parallel processing

sebkopf, 2019-04-12:
parallel processing can jumble the order of the files that were read. this is not usually a problem but can lead to confusion for users when file info is aggregated seemingly out of order.

japhir, 2019-05-01:
I think the below errors may be related to this fix:

```{r}
Warning messages:
1: All elements of `...` must be named.
Did you want `data = c(entry)`? 
2: All elements of `...` must be named.
Did you want `data = c(root, path, file_n, files_n, cachepath, reader_options, extension, 
    type, call, func, cacheable, description, env, read_from_cache, 
    write_to_cache, write_to_cache_if_errors)`? 
```

reading in both did and caf files seems to work fine, but it always throws these errors.

japhir, 2019-05-01:
also getting this warning while installing dev branch with newest rlang:
```
Warning: Unquoting language objects with `!!!` is deprecated as of rlang 0.4.0.
Please use `!!` instead.

  # Bad:
  dplyr::select(data, !!!enquo(x))

  # Good:
  dplyr::select(data, !!enquo(x))    # Unquote single quosure
  dplyr::select(data, !!!enquos(x))  # Splice list of quosures

This warning is displayed once per session.
```
didn't know where to drop these, since they seem minor and creating a new issue could have been excessive.

sebkopf, 2019-07-01:
@japhir I think this are fixed in dev, could you test when you get the chance and see if you still get those warning messages? Thanks!

japhir, 2019-07-10:
@sebkopf nice! Did you also see the new `{{` 'curly-curly' quote notation available in [rlang 0.4.0](https://www.tidyverse.org/articles/2019/06/rlang-0-4-0/)? That should be very nice and get rid of all the `param <- enquo(param)` calls at the heads of all functions.

japhir, 2019-07-10:
In the new dev I still have the first issue above: https://github.com/isoverse/isoreader/issues/65#issuecomment-488235254.

sebkopf, 2019-07-10:
strange, thanks for testing! and the `{{` note :)
could you send me a short file and with code example to make sure I'm trying to reproduce the same way?

japhir, 2019-07-11:
```{r}
library(tidyverse)
library(plotly)  # for interactive graphs
library(glue)
library(isoreader)
library(clumpedr)
```
reading the data from raw files triggers the bug:
```{r}
folderstr <- "/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData"
dids <- iso_read_dual_inlet(glue("{folderstr}/253pluskiel/Raw Data/Kiel Raw Data"),
                            cache = TRUE,
                            discard_duplicates = FALSE,
                            parallel = TRUE)
```
results in:
```
Warning: some files from different folders have identical file names:
	/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData/253pluskiel/Raw Data/Kiel Raw Data/_180316_4/180316_4_Std test_1_ETH-1(1).did
	/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData/253pluskiel/Raw Data/Kiel Raw Data/_180316_4/180316_4_Std test_2_ETH-1.did
	/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData/253pluskiel/Raw Data/Kiel Raw Data/_180316_4/180316_4_Std test_3_ETH-2.did
	/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData/253pluskiel/Raw Data/Kiel Raw Data/_180316_4/180316_4_Std test_4_ETH-2.did
	/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData/253pluskiel/Raw Data/Kiel Raw Data/_180316_4/180316_4_Std test_5_ETH-3.did
	/run/user/1000/gvfs/smb-share:server=geofile02.geo.uu.nl,share=geo-labs/@RawData/253pluskiel/Raw Data/Kiel Raw Data/_180316_4/180316_4_Std test_6_ETH- [... truncated]
Info: preparing to read 9637 data files (all will be cached), setting up 4 ...
Info (process 1): reading file '_180223_1/180223_1_Kiel Std test_1_ETH-1.di...
Info (process 1): reading file '_180228_1/180228_1_Kiel Std test_6_ETH-3.di...
[... omitted for brevity...]
Warning (process 3): caught error - cannot locate voltage data - block 'CTw...
[... omitted for brevity...]
Info (process 4): reading file 'Kiel Raw Data_180316_4/180317_4_Std test_17...
Info (process 1): reading file 'Kiel Raw Data_180316_4/180317_4_Std test_18...
Info (process 2): reading file 'Kiel Raw Data_180316_4/180317_4_Std test_19...
Info (process 3): reading file 'Kiel Raw Data_180316_4/180317_4_Std test_20...
Progress: [============================================>] 9584/9637 ( 99%)  2mError in base::nchar(strip_style(x), ...) : 
  invalid multibyte string, element 1
In addition: Warning messages:
1: All elements of `...` must be named.
Did you want `data = c(entry)`? 
2: All elements of `...` must be named.
Did you want `data = c(root, path, file_n, files_n, cachepath, reader_options, extension, 
    type, call, func, cacheable, description, env, read_from_cache, 
    write_to_cache, write_to_cache_if_errors)`? 
```
So this results in an an error sometimes. I can just re-run and then mostly it'll work.

But reading from the saved data storage also triggers the error:
```{r}
iso_save(dids, "out/dids.di.rds")
dids <- iso_read_dual_inlet("out/dids.di.rds")
```
results in:
```
Registered S3 method overwritten by 'R.oo':
  method        from       
  throw.default R.methodsS3
Info: preparing to read 1 data files (all will be cached)...
Info: reading file 'out/dids.di.rds' with '.di.rds' reader
Info: loaded data for 9602 data files from R Data Storage - checking loaded...
Progress: [===================================================] 1/1 (100%)  7s
Info: finished reading 1 files in 6.87 secs
Info: encountered 1608 problems in total
# A tibble: 1,608 x 4
   file_id             type  func            details                            
 * <chr>               <chr> <chr>           <chr>                              
 1 180223_1_Kiel Std … error extract_did_ra… cannot locate voltage data - block…
 2 180223_1_Kiel Std … error extract_did_ve… cannot process vendor computed dat…
 3 180306_2_Kiel Std … error extract_did_ra… cannot locate voltage data - block…
 4 180306_2_Kiel Std … error extract_did_ve… cannot process vendor computed dat…
 5 180307_2_Kiel Std … error extract_did_ra… cannot locate voltage data - block…
 6 180307_2_Kiel Std … error extract_did_ve… cannot process vendor computed dat…
 7 180308_3_STDs_12_E… error extract_isodat… "cannot process measurement info -…
 8 180314_3_Kiel Std … error extract_did_ra… cannot locate voltage data - block…
 9 180314_3_Kiel Std … error extract_did_ve… cannot process vendor computed dat…
10 180315_4_Std test_… error extract_did_ra… cannot locate voltage data - block…
# … with 1,598 more rows

Warning message:
All elements of `...` must be named.
Did you want `data = c(root, path, file_n, files_n, cachepath, reader_options, extension, 
    type, call, func, cacheable, description, env, read_from_cache, 
    write_to_cache, write_to_cache_if_errors)`? 
```

sebkopf, 2019-07-14:
very strange, can still not reproduce. Could you send me the `out/dids.di.rds` (offline) or some subset of it? or maybe can you run it with `options(warn = 2)` set prior to the command so the warning is turned into an error and it gives you a stacktrace to (hopefully) pinpoint where this error is thrown.

closed 2019-10-18 by sebkopf

sebkopf, 2019-11-01:
Totally understand, and I think I've found the problem when I was working on tidyr 1.0 integration a little while ago, it's the `nest` statements that cause this warning but should be fixed with the change to `nest(data = c(-v)) %>%` in utils.R (that's where the `data = c(entry)` suggestion comes from, and likewise the `nest(data = c(-process)) %>% ` in isoread.R

But this means it should be fixed in the newest isoreader version (1.0.11), do you still get the warning with that one?

japhir, 2019-11-12:
The error seems gone in the last version! :)