isoverse/isoreader #66: iso_get_raw_data hangs when reading in many files on latest release!

japhir, 2019-05-01:
On dev when I try to get the raw data from ~8k files, the function never completes! It gives the info message, but then never finishes without any errors. Have to interrupt with <kbd>Ctrl</kbd><kbd>c</kbd>. When I first `iso_filter_files()` to reduce the total number a bit, it doesn't hang.

japhir, 2019-05-01:
On ~4k caf files it runs, but takes a lot longer than before.

japhir, 2019-05-01:
OK, so on the 8k one it doesn't hang, it has just become much more slow for some reason! I let it finish, and now it took ~5 minutes!

sebkopf, 2019-05-01:
mh, nothing has changed about the `iso_get_raw_data` but the `iso_get_file_info` got an upgrade for safety checks on the file info data column types which I suspect could be the real culprit for this many data files. could you do me a favour and run a quick check on the two? probably easiest to tictoc it. Maybe just with the 4k data set so it doesn't take forever :) thanks for testing!

```r
library(tictoc)
tic()
iso_get_raw_data(...., include_file_info = NULL)
toc()
tic()
is_get_file_info(..., select = everything())
toc()
tic()
iso_get_raw_data(..., include_file_info = everyting())
toc()
```

japhir, 2019-05-02:
```{r}
> library(tictoc)
> tic()
> iso_get_raw_data(cafs, include_file_info = NULL)

Info: aggregating raw data from 4928 data file(s)
# A tibble: 74,674 x 9
   file_id                type   cycle v44.mV v45.mV v46.mV v47.mV v48.mV v49.mV
   <chr>                  <chr>  <int>  <dbl>  <dbl>  <dbl>  <dbl>  <dbl>  <dbl>
 1 170126_170124_Sibren_… stand…     0 13091. 15594. 18622.  2078.   216. -0.528
 2 170126_170124_Sibren_… stand…     1 12177. 14506. 17323.  1933.   201. -0.503
 3 170126_170124_Sibren_… stand…     2 11329. 13497. 16119.  1799.   187. -0.456
 4 170126_170124_Sibren_… stand…     3 10556. 12576. 15019.  1677.   174. -0.431
 5 170126_170124_Sibren_… stand…     4  9845. 11729. 14008.  1564.   163. -0.397
 6 170126_170124_Sibren_… stand…     5  9192. 10952. 13080.  1461.   152. -0.363
 7 170126_170124_Sibren_… stand…     6  8591. 10236. 12224.  1366.   142. -0.339
 8 170126_170124_Sibren_… stand…     7  8034.  9572. 11431.  1278.   133. -0.308
 9 170126_170124_Sibren_… stand…     8  7521.  8961. 10702.  1197.   124. -0.282
10 170126_170124_Sibren_… sample     1 12661. 14953. 17854.  1974.   206. -0.509
# … with 74,664 more rows
> toc()

10.496 sec elapsed
> tic()
> iso_get_file_info(cafs, select = everything())

Info: aggregating file info from 4928 data file(s)
# A tibble: 4,928 x 21
   file_id file_root file_path file_subpath file_datetime       Line 
   <chr>   <chr>     <chr>     <chr>        <dttm>              <chr>
 1 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 2 170126… /run/use… 170127_S… NA           2017-01-26 23:02:29 7    
 3 170127… /run/use… 170127_S… NA           2017-01-27 01:30:57 11   
 4 170127… /run/use… 170127_S… NA           2017-01-27 03:52:28 15   
 5 170127… /run/use… 170127_S… NA           2017-01-27 05:50:08 19   
 6 170127… /run/use… 170127_S… NA           2017-01-27 08:08:27 23   
 7 170127… /run/use… 170127_S… NA           2017-01-27 16:58:16 1    
 8 170127… /run/use… 170127_S… NA           2017-01-27 19:55:21 4    
 9 170127… /run/use… 170127_S… NA           2017-01-27 22:19:21 8    
10 170128… /run/use… 170127_S… NA           2017-01-28 00:49:42 12   
# … with 4,918 more rows, and 15 more variables: `Peak Center` <chr>,
#   Pressadjust <chr>, Background <chr>, `Reference Refill` <chr>, `Weight
#   [mg]` <chr>, Sample <chr>, `Identifier 1` <chr>, `Identifier 2` <chr>,
#   Analysis <chr>, Comment <chr>, Preparation <chr>, `Pre Script` <chr>, `Post
#   Script` <chr>, Method <chr>, MS_integration_time.s <dbl>
> toc()

2.245 sec elapsed
> tic()
> iso_get_raw_data(cafs, include_file_info = everything())

Info: aggregating raw data from 4928 data file(s), including file info 'everything()'
# A tibble: 74,674 x 29
   file_id file_root file_path file_subpath file_datetime       Line 
   <chr>   <chr>     <chr>     <chr>        <dttm>              <chr>
 1 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 2 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 3 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 4 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 5 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 6 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 7 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 8 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
 9 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
10 170126… /run/use… 170127_S… NA           2017-01-26 20:29:47 3    
# … with 74,664 more rows, and 23 more variables: `Peak Center` <chr>,
#   Pressadjust <chr>, Background <chr>, `Reference Refill` <chr>, `Weight
#   [mg]` <chr>, Sample <chr>, `Identifier 1` <chr>, `Identifier 2` <chr>,
#   Analysis <chr>, Comment <chr>, Preparation <chr>, `Pre Script` <chr>, `Post
#   Script` <chr>, Method <chr>, MS_integration_time.s <dbl>, type <chr>,
#   cycle <int>, v44.mV <dbl>, v45.mV <dbl>, v46.mV <dbl>, v47.mV <dbl>,
#   v48.mV <dbl>, v49.mV <dbl>
> toc()

11.899 sec elapsed
```

japhir, 2019-05-02:
I didn't benchmark anything before, but it felt pretty instantaneous before, and now I have to wait a while.

japhir, 2019-05-02:
I'm checking now whether it goes faster if i pre-select the columns of interest. For my [`get_inits()`](https://github.com/isoverse/clumpedr/blob/4cc0f00a9f827db8aa7200d9f56c52a972074fce/R/clean_did_info.R#L58) function I'm getting the raw data of only mass 44 and filtering the first rows, then appending it to my info tibble

closed 2020-03-01 by sebkopf