isoverse/isoreader #69: question about installation

BrianEads, 2019-05-16:
Hello, thanks for this fantastic package! It works great (when I use it on a machine on a network, where I can do devtools::github_install() to install).  When I try to install onto an older (Win XP) computer that cannot be on a network (long story) using the .zip file I downloaded from this repo, the command does not fail or error, but the package does not get installed. It seems like the issue might be that installing the software requires concomitant installation of other software dependencies, and these are thwarted by the fact that the computer isn't on the network?  Any idea if this is true?

sebkopf, 2019-05-17:
Hi @BrianEads, this is a problem that's come up in the past with offline IRMS systems so we started writing some documentation for it but haven't posted it yet. I'm pasting the current draft below and would very much welcome any feedback on whether it works and if not where it fails. Eventually we'll add this to the main `README.md` so it's easy to find.

#### Installing on a system not connected to the internet

IRMS instrumentation is sometimes kept offline to prevent problems with automatic Windows updates and viruses. This requires an offline installation of `isoreader` to make it possible to use the package directly on the instrument computer (which is typically an older Windows machine).
 
For this to work, you need to download all the packages that isoreader depends on as well as all packages that those dependencies depend on which can get pretty elaborate pretty quickly. The easiest way to make this possible is thus by just pulling *all* current release packages from CRAN and transferring them to the offline system (warnings, these are a *lot* of files, several Gb total, and may take a long time to download). The CRAN directory depends on the operating system, R version (must match!) and source vs. binaries. For example, to download all the R 3.5.x windows binaries, run in the following in a terminal (Note: these are unix/MacOS syntax, not sure exactly about the Windows equivalent):

```coffee
mkdir CRAN_packages
cd CRAN_packages
wget -nc ftp://cran.r-project.org/pub/R/bin/windows/contrib/3.5/*.zip
```

As well as the package(s) hosted on GitHub (run in terminal within CRAN_packages folder), which need to be converted to tar balls for installation (zip files don't work with ```install.packages```):
```coffee
mkdir GitHub_packages
cd GitHub_packages
curl -L -o tmp.zip https://github.com/isoverse/isoreader/archive/master.zip && unzip tmp.zip && rm tmp.zip
```

Transfer all these files to the offline system e.g. via USB stick, specify the `source_path`, then generate a package dependency tree (platform dependent) from an RStudio command line or R console, and install the GitHub packages from the sources files.

**Dependency tree:**

```coffee
source_path <- "C:/path/to/CRAN_packages" # windows
#source_path <- "/path/to/CRAN_packages" # unix
library(tools)
write_PACKAGES(source_path, type = "win.binary")
#write_PACKAGES(source_path, type = "mac.binary")
```

**Installation**:

```coffee
# local repository
url <- paste0("file:", source_path) # windows
#url <- paste0("file:\\", source_path) # unix

# install devtools
install.packages("devtools", contriburl = url)

# install github package(s) with dependencies
sapply(c("isoreader"), function(i) {
  pkg_path <- file.path(source_path, "GitHub_packages", paste0(i, "-master"))
  stopifnot(file.exists(pkg_path))
  deps <- devtools::dev_package_deps(pkg_path, dep = T)$package
  install.packages(deps, contriburl = url)
  tryCatch(devtools::install(pkg_path), error = function(e) message("NOTE: ", e$message))
  return("")
})
```

Note that sometimes Rgui.exe on windows will throw an error along the lines of *"The procedure entry point ... could not be located in the dynamic library ..."* when trying to load the newly installed packages during installation. This can be safely ignored and packages that otherwise installed without trouble should load okay when loading them with `library()`.

#### Updating a system not connected to the internet

If you're just updating an existing offline installation with a new version of any of the packages (simplest case: a new version of **isoreader**), download the latest version of the package from [github.com/isoverse/isoreader/archive/master.zip]([https://github.com/isoverse/isoreader/archive/master.zip), unzip the archive and transfer the ```isoreader-master``` folder to the offline system. Then start RStudio or an R console, set your working directory (```setwd("path_to_folder")```) to the folder where ```isoreader-master``` is located (not the ```isoreader-master``` folder itself, one folder above that), and run the following command. Adjust as needed for updating other packages.

```coffee
devtools::install("isoreader-master", reload = FALSE)
```

sebkopf, 2019-05-17:
note for dev team: we may be able to provide an in-package function (with OS an R version params) at some point that generates a bundle for offline installation (including an R script that installs everything once executed on the offline system) using the recursive dependency retrieval function of the [miniCRAN](https://cran.r-project.org/web/packages/miniCRAN/vignettes/miniCRAN-dependency-graph.html) package. This would make it a lot easier for users with offline IRMS systems.

BrianEads, 2019-05-17:
thanks so much for the prompt and VERY detailed reply!  I really appreciate your efforts.  If we do decide to go down the route of installing on the non-networked computer, I will keep you posted - it does have an open USB so we can copy the data off to another (networked) machine instead.

closed 2019-05-17 by BrianEads