isoverse/isoreader #70: Iso_reader has an invalid format error

cubessil, 2019-06-28:
When running isoreader on a set of cached data it will some times give the bellow error. Restarting R is normally enough to fix it. This error was created on a windows 10 computer.

Info: preparing to read 2313 data files (all will be cached), setting up 12 parallel processes...                                                                                     
Progress: [--------------------------------------------------------------------------------------------------------------------------------------------------------] 0/2313 (  0%)  0s213 parsing failures.
row col           expected    actual                                                                         file
143  X1 delimiter or quote i         'C:\Users\CUBESSIL\AppData\Local\Temp\RtmpAJgEKL\file32dc60eccf6.log'
184  X3 delimiter or quote i         'C:\Users\CUBESSIL\AppData\Local\Temp\RtmpAJgEKL\file32dc60eccf6.log'
184  X3 delimiter or quote i         'C:\Users\CUBESSIL\AppData\Local\Temp\RtmpAJgEKL\file32dc60eccf6.log'
184  -- 3 columns          5 columns 'C:\Users\CUBESSIL\AppData\Local\Temp\RtmpAJgEKL\file32dc60eccf6.log'
201  -- 3 columns          1 columns 'C:\Users\CUBESSIL\AppData\Local\Temp\RtmpAJgEKL\file32dc60eccf6.log'
... ... .................. ......... ............................................................................
See problems(...) for more details.
Error in sprintf("Info (process %d): ", X2) : 
  invalid format '%d'; use format %s for character objects"

sebkopf, 2019-07-01:
I think this has to do with the many parallel processes with some finishing at the exact same time and corrupting the log file. Could you test whether this also happens in the dev branch? should be more robust now against log file corruption.

sebkopf, 2020-03-01:
can we close this @brettdavidheiser

closed 2020-09-29 by sebkopf