isoverse/isoreader #72: error reading dxf files with custom gas configuration name

sebkopf, 2019-07-02:


wetlandscapes, 2020-02-07:
Hi,
Just getting into it, but super cool package!
I'm submitting this here, in the issue I've identified is related. I can re-post as new issue, if not.

In essence I'm reading in a dxf file for water from a Thermo Flash system. However, I get the same error for all the files being read in using `iso_read_continuous_flow()`: "Warning: caught error - could not find gas configuration for gas 'H2', available: 'CO'".

Going through some of your unexported code I'm noticing that the related functions in [isoread_dxf.R](https://github.com/isoverse/isoreader/blob/daaca68932b834d118b6a1c33ae8fe4bb6c79f2b/R/isoread_dxf.R) may be important. Specifically, the for loop at around line 62. That is, that loop only ever seems to identify one configuration gas, even though we have two.

Attached is a zip of one of the files as an example: [water_sample.zip](https://github.com/isoverse/isoreader/files/4172985/water_sample.zip).

Please let me know if there is any additional info I can provide to try and help. I'm pretty useless with binary data (one of the reasons I was so excited about your package!), but hopefully there is some other way I might be able to help.

Best,

Jason

sebkopf, 2020-02-08:
Hi Jason,
Thank you for submitting the example file. This file has a slightly different structure for the gas configuration information that we've not encountered before. A more robust gas config finding function that catches both the H2 and CO configuration in this file is now in the dev branch. If you'd like to start using this ahead of the next merge into the master, please use the following code to install directly from the dev branch:

```
devtools::install_github("isoverse/isoreader", ref = "dev")
iso_files <- iso_read_continuous_flow("water_sample.dxf", read_cache = FALSE)
```

closed 2020-02-08 by sebkopf

wetlandscapes, 2020-02-08:
Worked like a charm! Thanks for the _very_ quick fix!

commit 2de54788 (2019-07-02): fix problem with custom named gas configs in .dxf (#72)

fixes #72
commit c1ebaa20 (2020-02-08): make finding gas configurations in dxf files more robust

closes #72