isoverse/isoreader #98: Windows 10 reading problem

cubessil, 2020-02-14:
Running windows 10 with everything updated (R, RStudio, and Packages) as of 2020/02/13 It looks to put 5 errors for every file. This is true for all files including ones that worked it the past. I attached a few files but any files should work.

error | extract_isodat_sequence_line_info
error | extract_isodat_measurement_info
error | extract_did_raw_voltage_data
error | extract_isodat_reference_values
error | extract_did_vendor_data_table

[37956__all$ETH2_12_195.zip](https://github.com/isoverse/isoreader/files/4205415/37956__all.ETH2_12_195.zip)

japhir, 2020-02-17:
just commenting to inform that when i try to read those files into my linux-based session with `isoreader::iso_read_dual_inlet("~/Downloads/brett_probs/")` I get correctly read-in files.

sebkopf, 2020-02-17:
thank you both for checking this, something changed about the binary file processing on windows and I could not yet figure out what it is

sebkopf, 2020-02-18:
commit 539b8ea5f15108f2e70f6fde11f7f9bc03c2ddd0 addresses the original error, commit 72e02e9 addresses another windows vs. unix issues in scan files

closed 2020-02-18 by sebkopf