isoverse/isoreader #144: JOSS Automated tests

kdorheim, 2020-12-04:
Related to: openjournals/joss-reviews/issues/2878

Isoreader does have automated testing that is deployed continuously via github actions 👏. Locally when I run devtools::test()

```
OK:       1241
Failed:   0
Warnings: 11
Skipped:  0
```
there are several warnings that were triggered, none of these seem to be alarming more of an artifact related to the way the tests were set up using random numbers. Could you confirm that none of these warnings are trival? Would it make sense to add documentation or suppress the warnings?

sebkopf, 2020-12-14:
Some of the warnings may be version related, especially with the [change in `sample` from R 3.6 onward](https://statisticaloddsandends.wordpress.com/2019/06/19/rs-sample-function-works-differently-from-r-3-6-0-onwards/). With `renv` set up now, would you be willing to run the following again to test?

1. `git clone --single-branch --branch joss https://github.com/isoverse/isoreader.git` 
2. `renv::restore()`
3. `devtools::test()`

With this, the only warning should be the following if running interactively in RStudio:

**test-continuous-flow.R:117: warning: test that dxf files can be read
[ONE-TIME WARNING] Forked processing ('multicore') is disabled in future (>= 1.13.0) when running R from RStudio, because it is considered unstable. Because of this, plan("multicore") will fall back to plan("sequential"), and plan("multiprocess") will fall back to plan("multisession") - not plan("multicore") as in the past. For more details, how to control forked processing or not, and how to silence this warning in future R sessions, see ?future::supportsMulticore**

kdorheim, 2020-12-15:
It looks like might be some information about the `Bioconductor` package may be missing

```
renv::restore()
Do you want to proceed? [y/N]: y
Retrieving 'https://mran.microsoft.com/snapshot/2020-04-24/bin/macosx/el-capitan/contrib/3.6/codetools_0.2-16.tgz' ...
	OK [file is up to date]
Error: failed to determine Bioconductor repositories
```

sebkopf, 2020-12-16:
Thank you for testing this. I'm not quite sure how to troubleshoot this error as the only attribute of the project's renv configuration that plays a role is the Bioconductor version specified in the [`renv.lock` lockfile (3.12)](https://github.com/isoverse/isoreader/blob/joss/renv.lock#L12), which is the correct version for the lock file's R version (4.0.2). The error is thrown by [the function that tries to find the BioConductor repository URLs](https://github.com/rstudio/renv/blob/master/R/bioconductor.R#L48) for this Bioconductor version which are normally just provided by the `BiocManager` package. Could you try `BiocManager::repositories()` and `BiocManager::repositories(version = "3.12")` and see if it provides repo URLs or a more informative error? I suspect this might be an R - BioConductor version conflict which would be unfortunate because it would require making the minimal R version for this package R 4.0 instead of 3.6

kdorheim, 2020-12-16:
Yes updating the R version 4 did the trick, which as mentioned is unfortunate. 

devtools::test() passes with 71 warnings 

```
OK:       1241
Failed:   0
Warnings: 71
Skipped:  0
```
all of the warnings look like they are related to 

```
test-scan.R:53: warning: test that scn files can be read
Problem with `mutate()` input `result`.
ℹ Unrecognized time zone 'America/New_York'
ℹ Input `result` is `purrr::map2(...)`.
```
sorry didn't mean to make this more complicated than it already was 😬

sebkopf, 2020-12-17:
okay, R 4.0 it is, probably better for bioconductor support in the long run, several of the rhdf related packages are not backwards compatible.

As for the time zone warnings, seems like a problem with some of the lubridate calls but I could not reproduce it setting my system time zone to `America/New_York`. Perusing stack exchange there is some suggestion this could be a problem in lubridate that occurs in some operating system / R version combinations. Would you mind sharing your `sessionInfo()`? I can try to reproduce the warnings to resolve or at least catch them in a more informative way.

kdorheim, 2020-12-22:
Hi @sebkopf here is the session info for you 

```
R version 4.0.3 (2020-10-10)
Platform: x86_64-apple-darwin17.0 (64-bit)
Running under: macOS Mojave 10.14.6

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
LAPACK: /Library/Frameworks/R.framework/Versions/4.0/Resources/lib/libRlapack.dylib

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] stats     graphics  grDevices datasets  utils     methods   base     

other attached packages:
[1] feather_0.3.5   readxl_1.3.1    isoreader_1.2.7 testthat_2.3.2 

loaded via a namespace (and not attached):
 [1] Rcpp_1.0.5          lubridate_1.7.9     tidyr_1.1.1         listenv_0.8.0       prettyunits_1.1.1  
 [6] ps_1.3.4            utf8_1.1.4          assertthat_0.2.1    rprojroot_1.3-2     digest_0.6.25      
[11] R6_2.4.1            cellranger_1.1.0    backports_1.1.9     pillar_1.4.6        rlang_0.4.7        
[16] progress_1.2.2      rstudioapi_0.11     callr_3.5.1         R.utils_2.9.2       R.oo_1.23.0        
[21] desc_1.2.0          devtools_2.3.2      readr_1.4.0         stringr_1.4.0       compiler_4.0.3     
[26] UNF_2.0.6           pkgconfig_2.0.3     base64enc_0.1-3     pkgbuild_1.1.0      globals_0.12.5     
[31] tidyselect_1.1.0    tibble_3.0.3        codetools_0.2-16    fansi_0.4.1         future_1.18.0      
[36] crayon_1.3.4        dplyr_1.0.2         withr_2.3.0         rhdf5filters_1.2.0  R.methodsS3_1.8.0  
[41] lifecycle_0.2.0     magrittr_1.5        zip_2.1.0           cli_2.0.2           stringi_1.4.6      
[46] renv_0.12.3         fs_1.4.2            remotes_2.2.0       xml2_1.3.2          ellipsis_0.3.1     
[51] generics_0.0.2      vctrs_0.3.4         openxlsx_4.1.5      Rhdf5lib_1.12.0     tools_4.0.3        
[56] glue_1.4.1          purrr_0.3.4         hms_0.5.3           processx_3.4.4      pkgload_1.1.0      
[61] parallel_4.0.3      yaml_2.2.1          rhdf5_2.34.0        BiocManager_1.30.10 sessioninfo_1.1.1  
[66] memoise_1.1.0       usethis_2.0.0    
```

Output from sessionInfo and devtools::test()
[sessionInfo.txt](https://github.com/isoverse/isoreader/files/5731762/sessionInfo.txt)
[devtools_test.txt](https://github.com/isoverse/isoreader/files/5731767/devtools_test.txt)

sebkopf, 2020-12-31:
thanks @kdorheim, sorry this took me a while to figure out. I installed Mojave and R 4.0.3 but could not reproduce the error, turns out it was something introduced in R 4.0.2 that crated issues for some MacOS systems with the timezone database R uses (apparently dependent on how R was installed). I could finally reproduce it by deliberately setting `Sys.setenv(TZDIR = "internal")`. There's a detailed and quite fascinating discussion in the lubridate issue tracker on this issue: https://github.com/tidyverse/lubridate/issues/928 

Anyways, long story short, it was fixed in the latest CRAN release of lubridate (1.7.9.2) so I updated the renv. Could you run the following one more time to test?

- `git clone --single-branch --branch joss https://github.com/isoverse/isoreader.git` or `git pull` if you still have the previous checkout
- `renv::restore()` (should updated `packageVersion("lubridate")` to 1.7.9.2)
- `devtools::test()`

Thank you!

sebkopf, 2021-01-13:
Hi @kdorheim - thanks again for all your work on this, any chance the lubridate update fixed the timezone warnings?

kdorheim, 2021-01-19:
@sebkopf sorry about the delay my availability has really become limited these days

no more time zone warning messages!!! This issue can be closed.

sebkopf, 2021-01-19:
wonderful thank you!

closed 2021-01-19 by sebkopf