isoverse/isoreader #152: Paper missing references

r-barnes, 2021-01-22:
The paper at https://github.com/isoverse/isoreader/blob/master/paper.md is missing references

openjournals/joss-reviews/issues/2878

sebkopf, 2021-02-16:
I'm really sorry @r-barnes, I could not figure out what references are missing. Did you mean that they are not included in the `paper.md` file? Is it okay to keep them in `paper.bib` at https://github.com/isoverse/isoreader/blob/master/paper.bib? The JOSS preview service incorporates the references from the bib file correctly, see https://whedon.theoj.org/preview?id=e12f09b279c4a7bd5d8a7f17 for the preview. Please kindly re-open if I missed what references you found missing.

closed 2021-02-16 by sebkopf

r-barnes, 2021-04-18:
I think there are human-readable ways of putting citations in markdown, see, for instance, https://rmarkdown.rstudio.com/authoring_bibliographies_and_citations.html. But let's forego that for this work.