isoverse/isoreader #168: 'iso_get_vendor_data_table' unable to select manually edited peak rows from .dxf files

niygooooo, 2021-06-10:
I am currently trying to export all data rows with assigned AAs based on pre-defined Rt's. The problem I am having is that even if the Rt for an manually edited row in the .dxf file has the same Rt as the machine selected peak, those edited rows are not shown in the .xls output likely because _Isodat_ names manually edited peaks with a "+" e.g. "23+" so my best guess is that the command **'iso_get_vendor_data_table'** does not understand if the peak 'Nr.' column is not a numeric value.

**Question:**
- is there a way to change the code to accept peak numbers as "23" and "23+" values?
- or can I change the "Nr." column into a text format to then avoid the issue between the above cases?

**My Code:**
`
rm(list=ls())
library(isoreader) # requires R version > 4
project_directory <- "C:/Users/..."

### Read the confinuous flow (.dxf) files from their project directory
cf_files <- iso_read_continuous_flow(project_directory)

### Referring to an AA Std chromatogram, define the expected RT for each AA e.g. based on average of previous runs
Ala_RT <- 1441
Gly_RT <- 1491
Thr_RT <- 1554
Ser_RT <- 1589
Val_RT <- 1628
Leu_RT <- 1735
Ile_RT <- 1758
Nle_RT <- 1810
Pro_RT <- 1972
Asp_RT <- 2058
Glu_RT <- 2240
Phe_RT <- 2313
Tyr_RT <- 2400
Lys_RT <- 2442

### Select the desired columns from the output table, and compile all of the output tables in a single dataframe
results_df <- cf_files %>% iso_get_vendor_data_table(select=c(
  "File_Name" = file_id, Nr., Start, Rt, End, 
  'Dilution' = 'Sample Dilution', 
  'Amp28' = 'Ampl 28', 
  'Amp29' = 'Ampl 29', 
  'BGD28' = 'BGD 28', 
  'BGD29' = 'BGD 29', 
  'Area All' = 'Intensity All', 
  'd15N' = 'd 15N/14N'
)) %>% iso_make_units_explicit() %>% base::data.frame()

### Clean up columns to reflect our Workpad structure
results_df$Sample_ID <- sub('.*__(.*?)\\.dxf*', '\\1', results_df$File_Name)                # Extract Sample ID from the File Name     
results_df$Width <- results_df$End..s.-results_df$Start..s.                                 # Calculate the Width (s) column
results_df$End..s. <- NULL                                                                  # Remove the End (s) column
results_df$AA <- 'blank'                                                                    # Add the AA column
results_df <- results_df[, c(1, 12, 14, 2, 3, 4, 13, 5, 6, 7, 8, 9, 10, 11)]                # Re-order the columns

### Assigning AA
results_df$AA <- ifelse(Ala_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Ala_RT+22, 'Ala',    
                 ifelse(Gly_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Gly_RT+22, 'Gly',    
                 ifelse(Thr_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Thr_RT+22, 'Thr',    
                 ifelse(Ser_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Ser_RT+22, 'Ser',    
                 ifelse(Val_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Val_RT+22, 'Val',    
                 ifelse(Leu_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Leu_RT+22, 'Leu',   
                 ifelse(Ile_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Ile_RT+22, 'Ile',    
                 ifelse(Nle_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Nle_RT+22, 'Nle',    
                 ifelse(Pro_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Pro_RT+22, 'Pro',    
                 ifelse(Asp_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Asp_RT+22, 'Asp', 
                 ifelse(Glu_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Glu_RT+22, 'Glu',  
                 ifelse(Phe_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Phe_RT+22, 'Phe',  
                 ifelse(Tyr_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Tyr_RT+22, 'Tyr',  
                 ifelse(Lys_RT-18 < results_df$Rt..s. & results_df$Rt..s. < Lys_RT+22, 'Lys',  NA))))))))))))))

### Remove the rows where AA = NA
results_df <- results_df[complete.cases(results_df), ]

### Export the results dataframe as a csv
setwd(project_directory)                                        # Sets the directory
write.csv(results_df,"workpad.csv", row.names = FALSE)          # Writes the file`

closed 2021-06-10 by niygooooo

sebkopf, 2021-06-12:
Hi @niygooooo : did you get this issue resolved? if not, please upload a few example files (as a .zip archive) so we can more easily reproduce the issue. 

Also, if you are mapping peaks by retention time, you might find the `iso_map_peaks` function in isoprocessor useful: https://isoprocessor.isoverse.org/reference/iso_map_peaks.html

That package is not fully on CRAN yet and not as well documented as isoreader but an example of how it can be used is available in the vignette for compound-specific analyses at https://isoprocessor.isoverse.org/articles/gc_irms_example_carbon.html#samples-and-standards-1 

Happy to provide some example code based on your files if you get stuck.

niygooooo, 2021-06-14:
Hi @sebkopf: yes I found the issue, Had to specify to only remove rows where the AAs = NA in the code below, otherwise the script removed my edited peak rows too.

`results_df <- results_df[complete.cases(results_df$AA), ]`

Thanks for your suggestion to use iso_map_peaks. I'll have a look at that to see if we can improve our script in the future!